es-ude/elastic-ai.runtime.enV5 #132: Stop using git submodules for third-party deps

glencoe, 2023-10-11:
I see two ways to fetch source code dependencies: git submodules vs [cmake fetchcontent](https://cmake.org/cmake/help/latest/module/FetchContent.html). Here a pro/con list for git submodules (some of the points might just be a matter of taste):

- contra submodules:
  - complicated to use and setup
  - can confuse IDEs
  - easy to accidentally pick a wrong commit
  - cloning takes forever (at least with current setup... can't we use smth like `--shallow` there?). Maybe silly, but i think this is bothering me the most.
  - verbose directory structure making it hard to tell which of the subfolders are just submodules and which ones actually belong to the current project
  - complicates finding the exact versions for each of the dependencies
- pro submodules:
  - easy to modify sources from dependencies
  - easy to push/publish changes for dependencies

So to summarize, I see why we'd want to use the submodules for dependencies, that we own, but I think we should be using cmake to fetch third-party deps.

DavidFederl, 2023-11-17:
Replace submodules with CMake fetchcontent.
While doing this replace FreeRTOS git dependency with download of LTS version from webpage.

DavidFederl, 2024-01-12:
Declate deps: [FetchContent_Declare](https://cmake.org/cmake/help/latest/module/FetchContent.html#command:fetchcontent_declare)
Load deps: [FetchContent_Populate](https://cmake.org/cmake/help/latest/module/FetchContent.html#command:fetchcontent_populate)