es-ude/elastic-ai.runtime.enV5 #147: Check Error when Downloading to Flash via HTTP

DavidFederl, 2023-10-25:


glencoe, 2023-10-31:
@HerrCooker As you've done #149 , can you comment on this issue?

@DavidFederl, @HerrCooker  I suggest we either check the python script the same way as in #149 or officially call it broken.

HerrCooker, 2023-10-31:
@DavidFederl and I have checked the readback from Vivado after flashing via HTTP, using several different files. All of the readbacks had no difference, when compared to the source files.

DavidFederl, 2023-10-31:
@glencoe: I would suggest to simply test the USB flash script to verify if it's working correctly. Because this should only take about 15 minutes. If it doesn't work correctly than we should flag it as broken and don't fix it for now.

HerrCooker, 2023-10-31:
@glencoe I flashed the fast blinking .bin file and the echo_server .bin file via USB and checked the readback files against the source files. Both times there were no differences.

closed 2023-11-02 by DavidFederl