es-ude/elastic-ai.runtime.enV5 #160: Check why the ADXL345b sensor can only be used with low frequency

DavidFederl, 2023-11-03:


DerDavidos, 2024-03-01:
With a bare metal system the maximal rate of 3200Hz is possible

telefonjoker100, 2024-03-28:
800Hz is maximum frequency for I2C. When using low-power mode it is 400Hz. The datasheet specifies that the I2C Baudrate needs to be at least Data-rate*500. 
The specified frequencies are available when using stream mode. Our implementation first checks a register, if new data are available. As soon as the flag is set, we read all the measurement data. The configuration of ADXL is also questionable. We need to change the implementation. But we should also change our init routine or give an alternative init routine. Therefore, we should abstract more and have low-level sensor functions and library functions that are likely to be used by the end-user.

This would include that the end user wants to do only a few things. I personally would not implement anything regarding auto-sleep mode and standby mode.
1. Init Sensor (Frequency, G-Value Range, Maybe Low-Power if possible, Auto-Sleep Mode) This could include a self-test. 
2. Read a specific number of values
3. Read for a specific time
4. Read continuous
5. Enable Standby Mode(Maybe not needed in general) 
6. Change Frequency
7. Change G-Value Range

Library should contain
1. Setup I2C with Baudrate. When the sensor is started the initial Datarate is 100Hz according to the datasheet. Our software could give us the same information. 
2. Enable/Disable Stream Mode
3. Set Sensor Data Frequency incl. Low-Power mode.
4. Set G-Value Sensor Range
6. Single Read with Data-Ready Flag(which we already have) 
7. Multiple Read with use of Stream-Mode and FIFO register
8. Test Selfmode
9. Enable/Disable Standby
10. Enable/Disable Auto-Sleep Mode

DavidFederl, 2024-04-27:
Solution for https://github.com/es-ude/elastic-ai.runtime.enV5/issues/160#issuecomment-2025453281 implemented with issue #216

closed 2024-04-27 by DavidFederl