es-ude/elastic-ai.runtime.enV5 #236: Set custom VID and PID for elasticNode. This way we can use normal librarys to detect the elastic node via usb-vendor.

DavidFederl, 2024-07-29:


DavidFederl, 2024-07-29:
@telefonjoker100 it would be nice if you can provide a little background, why we should need a feature like that.

telefonjoker100, 2024-07-29:
The name of the devices changes often and is different for the operating systems: mac, linux und windows. With the PID and VID it is possible to detect the device on each operating systems and connect to it. In addition, it changes also on my mac from time to time and it is annoying to always change the serial port, because it has a new name now.

telefonjoker100, 2024-08-02:
VID: [0x1209](https://pid.codes/1209/)
PID: 4C48

LeoBuron, 2024-08-16:
VID and PID got now accepted.

DavidFederl, 2024-08-17:
The VID for RaspberryPi Products is set to `0x2E8A`. For the bootsel stage (mass storage to upload uf2) is set to PID `0x0003`. For the serial CDC connection (serial print output) the PID is set to `0x000A`.

The PID for the bootsel stage is set during production and embedded into the pico bootloader stage 1 inside the RP2040 ROM. It can't be changed afterwards and is there fixed for ever.

The VID and PID for the serial CDC connection can be changed by modifying the TinyUSB library and can therefore be changed by us. (see edited task description)

glencoe, 2024-10-25:
This python function works at least on MacOS:

```python
def get_pico_port():
    for c in list_ports.comports():
        if c.vid == 0x2e8a and c.pid == 0x000A:
            return c.device
```