es-ude/elastic-ai.runtime.enV5 #280: Update clang format

DavidFederl, 2024-09-13:


glencoe, 2024-10-20:
Tbh, i'm quite skeptical of why we'd want to do this. Most important thing about formatting from my perspective is consistency. But i don't mind making the change on an occasion where no open feature branches exist.

DavidFederl, 2024-10-30:
> Tbh, i'm quite skeptical of why we'd want to do this. Most important thing about formatting from my perspective is consistency. But i don't mind making the change on an occasion where no open feature branches exist.

The change was intended to be done when no open feature branches exist, to prevent major merge conflicts and enforce consistent code style, because now this is not the case.