es-ude/elastic-ai.runtime.enV5 #289: Automated Integrations Tests for enV5

LeoBuron, 2024-09-27:
As a developer I want to run automated integration tests. 

- [ ] Move hardware "tests" to develop folder
- [ ] Create automated test for each feature in unity
- [ ] Integrate Valgrind for memory leak detection
- [ ] Update workflow documentation for pull requests

glencoe, 2024-10-16:
So... what are we missing? AFAIK the most important feature for this would be a way to force the env5 into bootloader mode via software and the ["usb" protocol](https://github.com/es-ude/elastic-ai.runtime.enV5/issues/245) that was implemented recently.
I somehow think i saw enabling the bootloader via software already. Did you show that, @HerrCooker?

glencoe, 2024-10-23:
To switch into booloader via software we can use the [pico_bootrom](https://www.raspberrypi.com/documentation/pico-sdk/runtime.html#group_pico_bootrom_1ga1bbe1c1df02daaa7b68a0c95fb9dc545) module from pico_runtime.